sparkfun/pdp-qwiic-micro-python #25: SparkFun Qwiic SerLCD

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun 16x2 SerLCD - RGB Text (Qwiic)](https://www.sparkfun.com/products/16397)
[SparkFun 16x2 SerLCD - RGB Backlight (Qwiic)](https://www.sparkfun.com/products/16396)
[SparkFun 20x4 SerLCD - RGB Backlight (Qwiic)](https://www.sparkfun.com/products/16398)

### Notes

[Arduino Library](https://github.com/sparkfun/SparkFun_SerLCD_Arduino_Library) (Not linked on product pages)
[Python Driver](https://github.com/sparkfun/Qwiic_SerLCD_Py) (Already exists, need to confirm it works in MicroPython and CircuitPython)

malcolm-sparkfun, 2024-11-11:
Tested on MicroPython and all the examples worked well. Only change I had to make to be compatible with MicroPython was removing "future" statements (in [this commit](https://github.com/sparkfun/Qwiic_SerLCD_Py/commit/2c51fcb38e6bf5643bc790e9da71cfba01715d9d)).

sfe-SparkFro, 2024-11-19:
Testing in CircuitPython, occasionally getting `OSError: [Errno 19] No such device` when I let some of the examples run for a while (eg. Example 9). I'm guessing it's probably an issue with my setup rather than any code, but figured I should mention it in case we can debug it.

closed 2024-11-20 by sfe-SparkFro