sparkfun/pdp-qwiic-micro-python #33: SparkFun Air Velocity Sensor Breakout

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun Air Velocity Sensor Breakout - FS3000-1005](https://www.sparkfun.com/products/18377)
[SparkFun Air Velocity Sensor Breakout - FS3000-1015](https://www.sparkfun.com/products/18768)

### Notes

[Arduino Library](https://github.com/sparkfun/SparkFun_FS3000_Arduino_Library)
[Python Driver] (placeholder)

malcolm-sparkfun, 2024-11-15:
Initial Python port of Arduino lib is here: [https://github.com/sparkfun/Qwiic_FS3000_Py/tree/v1.0](https://github.com/sparkfun/Qwiic_FS3000_Py/tree/v1.0). unvalidated

malcolm-sparkfun, 2024-12-03:
Tested on MicroPython with a 1005 version of board and example behaved similar to Arduino library.  Note: This leveraged the changes to qwiic_i2c from this branch:

[https://github.com/sparkfun/Qwiic_I2C_Py/tree/feature/add_no_addr_reads](https://github.com/sparkfun/Qwiic_I2C_Py/tree/feature/add_no_addr_reads)

Moving to review

closed 2025-05-08 by malcolm-sparkfun