sparkfun/pdp-qwiic-micro-python #34: SparkFun Qwiic MicroPressure Sensor

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun Qwiic MicroPressure Sensor](https://www.sparkfun.com/products/16476)

### Notes

[Arduino Library](https://github.com/sparkfun/SparkFun_MicroPressure_Arduino_Library)
[Python Driver] (placeholder)

malcolm-sparkfun, 2024-11-15:
Initial python port here: [https://github.com/sparkfun/Qwiic_MicroPressure_Py/tree/v1.0](https://github.com/sparkfun/Qwiic_MicroPressure_Py/tree/v1.0). Unvalidated on hardware.

Played with using Doxygen comments in the qwiic_micropython.py rather than doc strings. Also pushed a "doxygen" directory with some of the doxygen output. I'm guessing we'll want that to look a bit different, but its something to look at while we start exploring doxygen in Python.

malcolm-sparkfun, 2024-11-25:
Tested example, working on MicroPython

closed 2025-05-08 by malcolm-sparkfun