sparkfun/pdp-qwiic-micro-python #35: SparkFun Qwiic Mux Breakout - 8 Channel (TCA9548A)

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun Qwiic Mux Breakout - 8 Channel (TCA9548A)](https://www.sparkfun.com/products/16784)

### Notes

[Arduino Library](https://github.com/sparkfun/SparkFun_I2C_Mux_Arduino_Library)
[Python Driver](https://github.com/sparkfun/Qwiic_TCA9548A_Py)

malcolm-sparkfun, 2024-11-11:
Made some tweaks to get this to work on MicroPython on this branch: [https://github.com/sparkfun/Qwiic_TCA9548A_Py/tree/micropython_fixes](https://github.com/sparkfun/Qwiic_TCA9548A_Py/tree/micropython_fixes) 

Since this device doesn't have a register map but rather has a single value to be read over I2C, I added the ability to pass "None" as the command code to the i2C read functions in the MicroPython and CircuitPython libraries. Let me know your thoughts on this. Branch with the changes to Qwiic_I2C_Py is here: [https://github.com/sparkfun/Qwiic_I2C_Py/tree/feature/add_no_addr_reads](https://github.com/sparkfun/Qwiic_I2C_Py/tree/feature/add_no_addr_reads)

malcolm-sparkfun, 2024-11-13:
Added new arduino examples, validated both on hardware. Example 2 uses two distance sensors and is dependent on [Qwiic_VL53L1X_Py](https://github.com/sparkfun/Qwiic_VL53L1X_Py).  It uses the I2C driver function __i2c_rdwr__, which originally only existed for Linux so to get it to work with MicroPython, you have to use a version of qwiic_i2c that [includes this commit](https://github.com/sparkfun/Qwiic_I2C_Py/commit/1151575d9a38c5f9c53fcbd78b1c18c88d1ed565)

sfe-SparkFro, 2024-11-20:
I'd like to have another example that uses a simpler sensor that customers are more likely to actually have, like a BME280 or something.

malcolm-sparkfun, 2024-11-20:
We can definitely do that. I think the idea with the multiple-sensor example was to show the workflow for interfacing with multiple sensors with the same I2C address and would likely be fairly trivial for users to swap in for their specific sensor there. 

If its common for users to have a bunch of BME280s, or we are trying to drive sales towards that, or if that would be simpler for users to follow in the example I'd be happy to write another example up for this for the BME280 though.

closed 2025-05-08 by malcolm-sparkfun