sparkfun/pdp-qwiic-micro-python #36: u-blox GPS

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun GPS-RTK2 Board - ZED-F9P (Qwiic)](https://www.sparkfun.com/products/15136)
[SparkFun GPS-RTK Board - NEO-M8P-2 (Qwiic)](https://www.sparkfun.com/products/15005)
[SparkFun GPS Breakout - ZOE-M8Q (Qwiic)](https://www.sparkfun.com/products/15193)
[SparkFun GPS Breakout - Chip Antenna, SAM-M8Q (Qwiic)](https://www.sparkfun.com/products/15210)
[SparkFun GPS-RTK Dead Reckoning Breakout - ZED-F9R (Qwiic)](https://www.sparkfun.com/products/22693)
[SparkFun GPS-RTK Dead Reckoning pHAT for Raspberry Pi](https://www.sparkfun.com/products/21305)
[SparkFun GPS Breakout - NEO-M9N, Chip Antenna (Qwiic)](https://www.sparkfun.com/products/15733)

### Notes

[Arduino Library](https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library)
[Python Driver](https://github.com/sparkfun/Qwiic_Ublox_Gps_Py)

sfe-SparkFro, 2024-11-04:
I poked a bit through the existing Python driver. It's written assuming Linux, particularly the imports of [ublox_gps.py](https://github.com/sparkfun/Qwiic_Ublox_Gps_Py/blob/master/ublox_gps/ublox_gps.py). Likely needs refactoring to work in MicroPython and CircuitPython.

malcolm-sparkfun, 2024-12-18:
Yes this would be a major refactor. As far as I can tell, "Qwiic" might be a misnomer for the existing Python driver because it looks like it optionally implements SPI or UART but not I2C (besides having a variable for the i2c address). It was written assuming Linux and leverages threading which seems like it is not supported by CircuitPython and is minimally supported by MicroPython.

malcolm-sparkfun, 2025-01-09:
We are putting this on the back-burner for now. We might create an entirely different repo for this that is a more direct port of the Arduino library.