sparkfun/pdp-qwiic-micro-python #37: SparkFun Triad Spectroscopy Sensor - AS7265x (Qwiic)

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun Triad Spectroscopy Sensor - AS7265x (Qwiic)](https://www.sparkfun.com/products/15050)

### Notes

[Arduino Library](https://github.com/sparkfun/SparkFun_AS7265x_Arduino_Library)
[Python Driver] (placeholder)

sfe-SparkFro, 2024-11-04:
Is it possible to have a single Arduino library and single Python driver?

malcolm-sparkfun, 2024-12-09:
It would be possible to combine these, but might get fairly cluttered with many functions that are only able to be used by one device or the other. I'll start by porting both of these libraries to Python and then if we think it makes sense after, I'd be happy to work on combining them.

malcolm-sparkfun, 2024-12-09:
Added initial python port of the triad (as7265x) here: [https://github.com/sparkfun/Qwiic_AS7265x_Py/tree/v1.0](https://github.com/sparkfun/Qwiic_AS7265x_Py/tree/v1.0). unverified

malcolm-sparkfun, 2024-12-10:
Added initial python port of the other spectral sensor lib here: [https://github.com/sparkfun/Qwiic_AS726x_Py/tree/v1.0](https://github.com/sparkfun/Qwiic_AS726x_Py/tree/v1.0). unverified

malcolm-sparkfun, 2024-12-11:
Made fixes and tested both python ports on hardware. For the most part,  values were in the same ballpark as Arduino lib. For the triad lib, the second example of basic readings with LEDs did seem much higher on the Python than Arduino though

closed 2025-05-08 by malcolm-sparkfun