sparkfun/pdp-qwiic-micro-python #39: ToF Distance Sensors

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun Distance Sensor - 1.3 Meter, VL53L4CD (Qwiic)](https://www.sparkfun.com/products/18993)
[SparkFun Distance Sensor Breakout - 4 Meter, VL53L1X (Qwiic)](https://www.sparkfun.com/products/14722)

### Notes

[Arduino Library](https://github.com/sparkfun/SparkFun_VL53L1X_Arduino_Library)
[Python Driver](https://github.com/sparkfun/Qwiic_VL53L1X_Py)

Both product pages link to the same drivers listed above. Confirm both the Arduino and Python drivers work with both boards (and if so, clarify that in the readme files and such)

sfe-SparkFro, 2024-11-04:
I believe these sensors have 16-bit addresses, so [Qwiic_I2C_Py](https://github.com/sparkfun/Qwiic_I2C_Py) would need to be modified to enable that.

malcolm-sparkfun, 2025-01-23:
Fixed that driver addition for CircuitPython and tested all examples on CircuitPython. Had to slightly tweak examples with this commit: 10c52e270146dc13a0d158c0891ca25439b3c0a1

closed 2025-05-08 by malcolm-sparkfun