sparkfun/pdp-qwiic-micro-python #40: SparkFun Qwiic ToF Imager - VL53L5CX

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun Qwiic ToF Imager - VL53L5CX](https://www.sparkfun.com/products/18642)
[SparkFun Qwiic Mini ToF Imager - VL53L5CX](https://www.sparkfun.com/products/19013)

### Notes

[Arduino Library](https://github.com/sparkfun/SparkFun_VL53L5CX_Arduino_Library)
[Python Driver] (placeholder)

malcolm-sparkfun, 2024-11-15:
Initial python port is here: https://github.com/sparkfun/Qwiic_VL53L5CX_Py/tree/v1.0. Still in a rough state and unvalidated. 

We are putting this on hold for now while we have a larger conversation about how to deal with devices that receive firmware pushes over I2C.

The Aruindo library for the VL53L5CX ToF Imager contains some huge buffers: https://github.com/sparkfun/SparkFun_VL53L5CX_Arduino_Library/blob/main/src/vl53l5cx_buffers.h

They contain firmware, etc. They get used in their initialization routine: https://github.com/sparkfun/SparkFun_VL53L5CX_Arduino_Library/blob/main/src/vl53l5cx_api.cpp#L297

malcolm-sparkfun, 2024-12-17:
Made fixes and tested examples in micropython. Moving to review

closed 2025-05-08 by malcolm-sparkfun