sparkfun/pdp-qwiic-micro-python #42: SparkFun 9DoF IMU Breakout - ICM-20948 (Qwiic)

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun 9DoF IMU Breakout - ICM-20948 (Qwiic)](https://www.sparkfun.com/products/15335)

### Notes

[Arduino Library](https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary)
[Python Driver](https://github.com/sparkfun/Qwiic_9DoF_IMU_ICM20948_Py)

malcolm-sparkfun, 2024-11-25:
Tested on MicroPython and the existing example seems to be working well for this, so moving this to review. There is only one example, but it encompasses the functionality of examples 1 and 2 in the arduino lib. Arduino examples 3-5 use SPI or interrupt pins which as of yet we aren't supporting in the Qwiic_Py libs. 

The other examples (6-11) use the ICM's Digital Motion Processor (DMP). This is another instance where we have vendor code and vendor firmware buffers that would likely be fairly messy to port to Python. It seems like users do have a want for this functionality in the python driver though (see [this issue](https://github.com/sparkfun/Qwiic_9DoF_IMU_ICM20948_Py/issues/1)). 

@sfe-SparkFro let me know if it seems like we should work to add DMP support before moving this to done

malcolm-sparkfun, 2024-11-25:
Tested on MicroPython and the existing example seems to be working well for this, so moving this to review. There is only one example, but it encompasses the functionality of examples 1 and 2 in the arduino lib. Arduino examples 3-5 use SPI or interrupt pins which as of yet we aren't supporting in the Qwiic_Py libs. 

The other examples (6-11) use the ICM's Digital Motion Processor (DMP). This is another instance where we have vendor code and vendor firmware buffers that would likely be fairly messy to port to Python. It seems like users do have a want for this functionality in the python driver though (see issue 1 in the Python repo).

@sfe-SparkFro let me know if it seems like we should work to add DMP support before moving this to done

closed 2025-05-08 by malcolm-sparkfun