sparkfun/pdp-qwiic-micro-python #43: SparkFun Triple Axis Accelerometer Breakout - KX13x

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun Triple Axis Accelerometer Breakout - KX132 (Qwiic)](https://www.sparkfun.com/products/17871)
[SparkFun Triple Axis Accelerometer Breakout - KX134 (Qwiic)](https://www.sparkfun.com/products/17589)

### Notes

[Arduino Library](https://github.com/sparkfun/SparkFun_KX13X_Arduino_Library)
[Python Driver](https://github.com/sparkfun/Qwiic_KX13X_Py)

malcolm-sparkfun, 2024-11-11:
Had to do a major refactor for this to match functionality of Arduino library and examples. 

Refactor is on this branch: [https://github.com/sparkfun/Qwiic_KX13X_Py/tree/refactor/match_arduino](https://github.com/sparkfun/Qwiic_KX13X_Py/tree/refactor/match_arduino)

sfe-SparkFro, 2024-11-19:
Good to make it match the Arduino library, thanks for doing that! Though we also want to maintain backwards compatibility since this Python port already existed, so users who update will have stuff broken. Should be easy enough to resolve by adding back the old methods that just call the new methods.

For example, `accel_control()` got renamed to `enable_accel()`. Just add `def accel_control()` that calls `enable_accel()`

sfe-SparkFro, 2024-11-19:
Also, Example 4 name is `qwiic_kxx13x_ex4.py`, has 1 too many `x`'s

malcolm-sparkfun, 2024-11-20:
I have just addressed these with commit: [https://github.com/sparkfun/Qwiic_KX13X_Py/commit/b41b17b30464b6d11623ad2eb5b780369513610f](https://github.com/sparkfun/Qwiic_KX13X_Py/commit/b41b17b30464b6d11623ad2eb5b780369513610f)

closed 2025-05-08 by malcolm-sparkfun