sparkfun/pdp-qwiic-micro-python #44: SparkFun 6DoF IMU Breakout - BMI270 (Qwiic)

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun 6DoF IMU Breakout - BMI270 (Qwiic)](https://www.sparkfun.com/products/22397)

### Notes

[Arduino Library](https://github.com/sparkfun/SparkFun_BMI270_Arduino_Library)
[Python Driver] (placeholder)

malcolm-sparkfun, 2024-11-15:
Looking through the Arduino Library, it looks like it heavily leverages the Bosch API which is very large. It contains 13 source files, one of which is > 10,000 lines

sfe-SparkFro, 2024-11-15:
Ok, pushing this back to the backlog. We should discuss what to do about libraries that use vendor APIs.

@gigapod thoughts?

gigapod, 2024-11-15:
Skip it - we don't want to port vendor code. I would be complicated...

The only reason to do this would be if we have funding or a large opportunity behind it.