sparkfun/pdp-qwiic-micro-python #50: SparkFun Pulse Oximeter and Heart Rate Sensor - MAX30101 & MAX32664 (Qwiic)

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun Pulse Oximeter and Heart Rate Sensor - MAX30101 & MAX32664 (Qwiic)](https://www.sparkfun.com/products/15219)

### Notes

[Arduino Library](https://github.com/sparkfun/SparkFun_Bio_Sensor_Hub_Library)
[Python Driver] (placeholder)

malcolm-sparkfun, 2024-12-05:
According to the Arduino Lib, we need to write to the mfio and ~reset pin in the following order to put the MAX32664 into application mode while coming out of reset:

MFIO HIGH
~RESET LOW
delay
~RESET HIGH

How do we want to handle this for python? Should we develop a module similar to the qwiic_i2c module that abstracts setting pins for different platforms? I suppose it could also be possible to have the user manually perform the above sequence with jumpers after powering up, but that sounds clunky. We could also throw this one back on the backlog if we don't intend to support setting pins from these python drivers

malcolm-sparkfun, 2025-01-08:
In my own experimentation with a single sensor, it seems to enter application mode if both the MFIO and ~RESET pin are left unconnected. According to the datasheet, the ~RESET pin is internally pulled up, so we can likely get away with leaving that unconnected, but for the MFIO pin it says "This pin provides different functions: MFIO asserts low as an output when the sensor
hub needs to communicate with the host controller; the pin acts as an input and causes the sensor hub to enter bootloader mode if the MFIO pin is low during a reset". It's not clear if the MFIO pin is pulled up, so I might just have been lucky that the floating MFIO input was high during reset when I left it unconnected and it is still required. 

It doesn't seem like the Arduino lib uses those pins for anything besides ensuring the sensor doesn't enter bootloader mode when coming out of reset.

@sfe-SparkFro Do you have any thoughts on where we should go with this one for the Python driver due to its potential requirement on toggling external pins?

Some Possibilities:
- Banish this to backlog
- Have users pass in a "set_mfio_pin" function and a "set_reset_pin" function at object instantiation or in begin/init function call. It is up to them to develop these functions for their specific platform and then the driver will invoke them at the correct time.
- When using the Python lib, instruct users to place an external pull up resistor on MFIO so it is high when its an input during reset as needed but can still get driven either high or low when the MFIO acts as an output.

sfe-SparkFro, 2025-01-08:
Can you use a multimeter to try to detect the presence of an internal pullup resistor? If no pullup is present, then IMO we push this to backlog (we should really discuss the best way to have platform agnostic GPIO control at some point).

malcolm-sparkfun, 2025-01-08:
Doesn't seem like MFIO is pulled up, when I tie ~RESET to GND putting the sensor in reset, MFIO reads only on the order of 10's of mV. However, the first Arduino example is working consistently for this when I leave MFIO and ~RESET unconnected. I can push this to backlog for now and revisit later