sparkfun/pdp-qwiic-micro-python #60: SparkFun GPS Breakout - XA1110 (Qwiic)

sfe-SparkFro, 2024-11-04:
### Board

SparkFun GPS Breakout - XA1110 (Qwiic)

### Notes

[Arduino Library](https://github.com/sparkfun/SparkFun_I2C_GPS_Arduino_Library)
[Python Driver](https://github.com/sparkfun/Qwiic_Titan_Gps_Py)

malcolm-sparkfun, 2024-12-03:
The existing py driver leverages the pynmea2 package which users can't directly install with pip/mip/circup for circuitpython and micropython, but they can likely pull in the source directly from: 

[https://github.com/Knio/pynmea2](https://github.com/Knio/pynmea2)

Looks like it also has some RasPi specific clock stretching content, so we will have to rework a bit for MicroPython and CircuitPython

malcolm-sparkfun, 2024-12-26:
Even after removing many of the dependencies in pynmea2 on packages that are not supported for MicroPython, it still didn't work on MicroPython because it leverages some features of python itself (like metaclasses) that are not supported by MicroPython. Since we are only using a small subset of the functionality of pynmea2, I swapped it out with a different Python NMEA parser called [micropyGPS ](https://github.com/inmcm/micropyGPS) that is designed to be able to run on Micropython.  I started a branch [[here](https://github.com/sparkfun/Qwiic_Titan_Gps_Py/tree/refactor/micropython)](https://github.com/sparkfun/Qwiic_Titan_Gps_Py/tree/refactor/micropython) with that module swapped in for parsing and it seems to be working fairly well in initial testing.

closed 2025-05-08 by malcolm-sparkfun