sparkfun/pdp-qwiic-micro-python #66: SparkFun Pressure Sensor - BMP581 (Qwiic)

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun Pressure Sensor - BMP581 (Qwiic)](https://www.sparkfun.com/products/20170)
[SparkFun Micro Pressure Sensor - BMP581 (Qwiic)](https://www.sparkfun.com/products/20171)

### Notes

[Arduino Library](https://github.com/sparkfun/SparkFun_BMP581_Arduino_Library)
[Python Driver] (placeholder)

malcolm-sparkfun, 2024-12-05:
This Arduino lib also leverages a vendor API. Their API doesn't look incredibly long, let me know if we want to port this one.

malcolm-sparkfun, 2024-12-20:
Initial Python port of Arduino lib here: [https://github.com/sparkfun/Qwiic_BMP581_Py/tree/v1.0](https://github.com/sparkfun/Qwiic_BMP581_Py/tree/v1.0)

Tested examples on MicroPython. Moving to review.

closed 2025-05-08 by malcolm-sparkfun