sparkfun/pdp-qwiic-micro-python #69: SparkFun LiPo Fuel Gauge - MAX17048

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun LiPo Fuel Gauge](https://www.sparkfun.com/products/20680)
Many of our dev boards

### Notes

[Arduino Library](https://github.com/sparkfun/SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library)
[Python Driver] (placeholder)

malcolm-sparkfun, 2024-11-15:
Initial Python port of arduino library: [https://github.com/sparkfun/Qwiic_MAX1704X_Py/tree/v1.0](https://github.com/sparkfun/Qwiic_MAX1704X_Py/tree/v1.0). unvalidated.

malcolm-sparkfun, 2024-11-21:
Initial testing with a max 17043 seemed to behave similar to the arduino lib, but we should still test this with an actual lipo as well as test with a 17048 since it has the extended feature set

malcolm-sparkfun, 2025-01-09:
@sfe-SparkFro Any idea how to best test this with the 17048? Our product that breaks out the 17048 here: [https://www.sparkfun.com/qwiic-fuel-gauge-max17048.html#](https://www.sparkfun.com/qwiic-fuel-gauge-max17048.html#) seems to be out of stock and I am not sure if we are restocking them any time soon. I remember you mentioning that the 17048 is on several of our other boards. Is there a convenient way you can think of to test the python lib against the fuel guage that's integrated into one of these other boards?

sfe-SparkFro, 2025-01-09:
Can grab one of these: https://www.sparkfun.com/sparkfun-thing-plus-rp2350.html

closed 2025-05-08 by malcolm-sparkfun