sparkfun/pdp-qwiic-micro-python #71: SparkFun I2C DAC Breakout - MCP4725

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun I2C DAC Breakout - MCP4725](https://www.sparkfun.com/products/12918) (Should make a Qwiic version)

### Notes

[Example code](https://learn.sparkfun.com/tutorials/mcp4725-digital-to-analog-converter-hookup-guide#firmware) (should turn this into an Arduino library)
[Python Driver] (placeholder)

malcolm-sparkfun, 2024-12-03:
Added initial Arduino lib here to encompass functionality of existing example code snippet. Not yet validated: 

[https://github.com/sparkfun/SparkFun_MCP4725_Arduino_Library/tree/v1.0](https://github.com/sparkfun/SparkFun_MCP4725_Arduino_Library/tree/v1.0) 

Since this is the first one of these I've done in Arduino, I'm sure I'll have to make plenty of stylistic changes post-review. Also curious to know if this library is something well-suited to pull in the toolkit for or since it is simple/small and SPI isn't available if we should leave it as-is with just using "Wire" from Arduino directly. 

If we think it is valuable, I can add some/all of these additional examples. Possibly:
1) Waveform (exisiting)
2) Writing EEPROM
3) Reading data
4) Power Down Modes
5) Fast Mode Write

Hardware for this is on backorder, but once I get a chance to validate and fix this lib, I can make a python port of it.

malcolm-sparkfun, 2024-12-03:
Kirk has confirmed that we do indeed want to use the Toolkit here

malcolm-sparkfun, 2024-12-09:
Switched to using toolkit and added and tested more examples for the Arduino lib. Will make a Python port for this now

malcolm-sparkfun, 2024-12-09:
Added quick Python port of above Arduino lib here: [https://github.com/sparkfun/Qwiic_MCP4725_Py/tree/v1.0](https://github.com/sparkfun/Qwiic_MCP4725_Py/tree/v1.0)

closed 2025-05-08 by malcolm-sparkfun