sparkfun/pdp-qwiic-micro-python #75: SparkFun Digital Temperature Sensor - TMP1xx

sfe-SparkFro, 2024-11-04:
### Board

[SparkFun Digital Temperature Sensor Breakout - TMP102](https://www.sparkfun.com/products/13314)
[SparkFun Digital Temperature Sensor - TMP102 (Qwiic)](https://www.sparkfun.com/products/16304)
[SparkFun High Precision Temperature Sensor - TMP117 (Qwiic)](https://www.sparkfun.com/products/15805)

### Notes

[Arduino Library (TMP102)](https://github.com/sparkfun/SparkFun_TMP102_Arduino_Library)
[Arduino Library (TMP102)](https://github.com/sparkfun/SparkFun_TMP117_Arduino_Library)
[Python Driver] (placeholder)

Might be possible to combine Arduino libraries into single one, then port to Python

malcolm-sparkfun, 2024-12-02:
Looking at the datasheets for each: 

[https://cdn.sparkfun.com/assets/2/6/8/0/c/tmp102-Datasheet.pdf](https://cdn.sparkfun.com/assets/2/6/8/0/c/tmp102-Datasheet.pdf)
[https://cdn.sparkfun.com/assets/9/e/5/1/8/tmp117.pdf](https://cdn.sparkfun.com/assets/9/e/5/1/8/tmp117.pdf)

It looks like it might not make sense to combine these two libraries. Seems like they have a different set of registers, and those that are shared (e.g. the configuration register) have different fields

malcolm-sparkfun, 2024-12-05:
Looks like there is already a Python port for the 102: [https://github.com/sparkfun/Qwiic_TMP102_Py/blob/main/qwiic_tmp102.py](https://github.com/sparkfun/Qwiic_TMP102_Py/blob/main/qwiic_tmp102.py)

malcolm-sparkfun, 2024-12-06:
Added initial Python port for the 117 here: [https://github.com/sparkfun/Qwiic_TMP117_Py/tree/v1.0](https://github.com/sparkfun/Qwiic_TMP117_Py/tree/v1.0). unverified

malcolm-sparkfun, 2024-12-06:
Tested Python port for the 117 on MicroPython and examples behaved similar to arduino

malcolm-sparkfun, 2024-12-06:
The Python port for the 102 needed some tweaks to get working on MicroPython. I added them in a branch here: [https://github.com/sparkfun/Qwiic_TMP102_Py/tree/feature/micropython](https://github.com/sparkfun/Qwiic_TMP102_Py/tree/feature/micropython)

I tested the examples with MicroPython and they behaved similar to Ardiuno lib.

Moving this to review. (since the 102 and 117 are two separate repos, we might want to split this into two issues moving forward)

closed 2025-05-08 by malcolm-sparkfun