sparkfun/pdp-qwiic-micro-python #132: Add doxygen created documentation

gigapod, 2025-01-09:
This task includes the following:

- [ ] Add Doxygen header comments to the projects source files
- [ ] Use the Doxygen templates/look and feel from the SparkFun Toolkit
- [ ] Add documentation / overview for the driver 
- [ ] Add documentation of examples
- [ ] Include boilerplate - how to install documentation for each supported package manager
- [ ] Automate documentation builds

malcolm-sparkfun, 2025-01-14:
GitHub actions are disabled for this repo so I am unable to deploy the doxygen github pages

brhoff720, 2025-01-14:
@malcolm-sparkfun Try now.  SparkFun actions should be usable.

malcolm-sparkfun, 2025-01-14:
@brhoff720 Still no luck. Oddly, this is the only one of these qwiic_py repos that has this turned off and gives the attached error when I try to enable GitHub pages deployment from actions:

[image]

closed 2025-05-08 by malcolm-sparkfun