sparkfun/pdp-qwiic-micro-python #399: Create New Skus and Sparkle Pages for MicroPyhton Preloaded Products

brhoff720, 2025-02-25:
This is a specific product effort based on the product submittal https://github.com/sparkfun/pdp-product-roadmap/issues/151, specifically a version of our forthcoming IoT RedBoard - RP2350 with MicroPython installed.

We also will be doing  the IoT RedBoard - ESP32 with MicroPython installed.

To launch these products - the following is needed:

No new hardware - used what is in stock
New SKU for this board -- based of the current product
Modification of the build/test fixture software to flash the MicroPython firmware
The MicroPython binary should be sourced from the [official MicroPython builds ](https://micropython.org/download/?port=esp32).

Notes:

The IoT RedBoard RP2350 is currently not available and must be added to the builds (part of this effort)

See the parent issue (https://github.com/sparkfun/pdp-product-roadmap/issues/151 ) for motivation and details behind this product, including the launch goals and incentives

Current IoT Redboard RP2350 Sku - https://sparkle.sparkfun.com/sparkle/products/27708  Should be able to copy this

IoT Redboard ESP32 - https://sparkle.sparkfun.com/sparkle/products/19177

brhoff720, 2025-02-25:
@SFE-Pearce @SFEvanWalbridge Need some help creating a couple new skus.  I think copy is ok, as we may want to save the one time clone for later.  Details above.  Would like to see this done by next Friday.  If other high priorities, let me know so we can adjust appropriately.

SFE-Pearce, 2025-02-28:
> [@SFE-Pearce](https://github.com/SFE-Pearce) [@SFEvanWalbridge](https://github.com/SFEvanWalbridge) Need some help creating a couple new skus. I think copy is ok, as we may want to save the one time [...]

https://sparkle.sparkfun.com/sparkle/parts/30257#tab-attributes - SparkFun IoT RedBoard - ESP32 MicroPython Development Board 

Let me know if any changes need to be made, but I think I have everything correct there.